VoronDesign/Voron-Switchwire #12: Error on page 10 and 14 - Wrong extrusions marked

internallabworks, 2021-01-27:
On page 10, the uprights for Z is marked for use for the bottom cross extrusions.
And on page 14 the cross extrusions are marked for use on Z
Example on page 10:
[image]

Makes the bottom look funky wide :)
[image]

closed 2021-01-30 by john--

jakep82, 2021-05-25:
This issue was reintroduced in the manual that was published 19 days ago. I just built my frame and discovered it called for the wrong extrusions, requiring me to disassemble and start over.

raymondh2, 2021-05-25:
This issue has been resolved by #13

PR #13 (2021-01-30): Update Assembly_Manual_SW.pdf
Fixes #12 
Corrected callouts for some screws