VoronDesign/Voron-Switchwire #16: Inconsistencies between BOM and Sourcing Guide

digitalfrost, 2021-02-22:
There are inconsistencies between the sourcing guide https://vorondesign.com/sourcing_guide?model=VS and the BOM available on the  https://vorondesign.com/voron_switchwire page 

Under the **Misc** section in the sourcing guide there is
_BMG Extruder Hardware Kit_
In the downloadable BOM this is listed in the **Motion** section as
_BMG Extruder Componnets Kit_

Under the **Motion** section in the sourcing guide there is
_Key-bak Super48 13oz_
this is listed in the **Misc** section in the downloadable BOM as
_Key-Bak Super 48 Retractable Keychain_
(I think this description is clearer as to what the item is)

renjie-teo, 2022-04-02:
Hi I've come across the link to this issue and would like to chime in for the keybak if it could be named as such:
Key-bak Super48 36" 13oz
The name is the full name listed on amazon as well so it would be clear to voron builders that they are supposed to purchase the 36" 13oz version instead of getting the 48" 8oz version by accident (perhaps also provide a warning since many who were unaware made the same mistake)

Thank you!

nemgrea, 2023-03-01:
this has been fixed

closed 2023-03-01 by nemgrea