VoronDesign/Voron-Switchwire #43: Manual only shows Probe Signal Wire Connection, Website is also unclear

ChaosBlades, 2022-03-08:
The Switchwire manual shows the diode and signal wire connection but just shows the positive and negative probe wires going to nothing.
[image]

The website could be more clear as well.
https://docs.vorondesign.com/build/electrical/sw_miniE3_v20_wiring.html
**Says...**
Plug Probe GND and Signal (with BAT85 diode) in to Z-STOP
Connect Probe +V to VIN in terminal next to HB

**Sound Say...**
Plug Probe Ground (Blue Wire) to Z-STOP GND and Signal (Black Wire) with BAT85 diode (black band facing away from connector) in to Z-STOP PC2
Connect Probe Positive (Brown Wire) to POWER 12/24V (The one just to the Left of BED)

ChaosBlades, 2022-03-08:
Also, The Mainboard Breakout Board pictured on page 125, I can't find anything on this board. It isn't linked to, I cant find any reference to it anywhere, and I certainly can't find anywhere to buy it. All the Toolhead PCB listings for purchase only come with the Toolhead PCB. It also shows the Probe being connected from a 3-pin connector to a 2-pin connector via 1 Brown Wire. So it is not clear how this is supposed to be connected even if I could find anything on this mysterious board.
[image]

Onlywavesmdp, 2022-06-18:
I have the same problem, were you able to solve it?

T4KUUY4, 2025-01-23:
@ChaosBlades 
Hoy there, late update but its an update :D

The board you are seeing here is an LDO specific toolhead board. 
Looks like we forgot to add a tag to that page stating that. This is now fixed within the next manual update.

~Takuya

closed 2025-01-23 by T4KUUY4