VoronDesign/Voron-Switchwire #60: Purge line g code causing issue during print start.

Johnn201, 2023-07-22:
Using the switchwire skr mini e3 v3 config file,and superslicer with PRINT_START in the start gcode after print start having the Purge Line Gcode uncommented I get error:
Extruder not hot enough
7:15 AM
Move out of range: 5.000 5.000 3.000 [0.000]
7:15 AM
Move out of range: 5.000 5.000 3.000 [0.000]
If I comment it out it prints normally.
[klippy.log](https://github.com/VoronDesign/Voron-Switchwire/files/12136471/klippy.log)

Johnn201, 2023-07-23:
I have gotten rid of all but the error Extruder not hot enough by changing G0 X5 Y5 to G0 X20 Y20 F6000 but still get Extruder not hot enough unless I preheat the extruder, so still looking for a solution.

Johnn201, 2023-07-23:
[Scientist_Test_90mm1.txt](https://github.com/VoronDesign/Voron-Switchwire/files/12138885/Scientist_Test_90mm1.txt)

nemgrea, 2023-07-24:
add 

M109
M190 

to the beginning of the purge g code

Johnn201, 2023-07-25:
I'll try that thank you.

Johnn201, 2023-07-26:
The M109
       M190
Also did not fix the heating problem 
Should "Only custom start GCode be checked in superslicer printer settings?